morcibacsi/PSAVanCanBridge #4: Feature - low washing fluid level indicator

karolp1993, 2020-04-19:
If we use sensor like this: https://pl.aliexpress.com/item/32894861981.html

We can make a simple low level of washing fluid indicator. With simple circuit, we can connect signal from sensor to ESP32 gpio input and switches from 0V to 3.3V or vice versa. Then, we can show popup on display, and with using a buzzer, signal a popup. I'm working about it.

morcibacsi, 2020-04-22:
This would be awesome.

Chaozzmax, 2021-03-02:
Any upate about this?

karolp1993, 2021-03-02:
Unfortunately not. I've made first sketch for tests and.. sold a car.

morcibacsi, 2021-03-02:
Then I guess we can close this one as I don't really have the time for such features. If anyone feels the need of it then later it can be reopened, and can be implemented.

closed 2021-03-05 by morcibacsi