morcibacsi/PSAVanCanBridge #16: Have access to diag settings

morcibacsi, 2021-04-14:
It would be nice to have support for changing some of the parameters which are only accessible from the diagnostic interface. Currently that requires a car with a AEE2004 BSI (with CAN bus). It would be nice to set parameters on the display and the radio as well.
For starter these settings should be available:

1.  Some of the Type C screen comes with a separate type of control switch which is changeable from diag.
2.  The color screen for the 407 has a startup screen, which can be changed in the settings with Peugeot and Citroen screens.

Chaozzmax, 2023-01-19:
Hi peter
Is there any update for this enhancement?

morcibacsi, 2023-01-19:
Hi, no update on this. Unfortunately I don't have much time to work on the project.

morcibacsi, 2026-05-24:
Implemented  with a third party app by e058a5a623734f42dd27fe0dc91e659568e5bc17
See telecoding section in the readme: https://github.com/morcibacsi/PSAVanCanBridge#telecoding

closed 2026-05-24 by morcibacsi