morcibacsi/PSAVanCanBridge #17: Ability to work with RT4

najdanovicivan, 2021-05-22:
I'm wondering if it will possible to use RT4 in 206CC ?

morcibacsi, 2021-05-22:
In theory it should work, although I've never tested it.

droofy, 2021-05-23:
@najdanovicivan , how will you integrate the RT4 screen in the dashboard?

najdanovicivan, 2021-05-23:
@droofy Maybe make a 3D printed bezel. This is all just an idea. I am still unable to find RT4 anywhere

droofy, 2021-05-23:
you will have to integrate a large screen. It must be difficult to realize in 3D
https://i97.servimg.com/u/f97/19/00/58/59/img_1511.jpg

morcibacsi, 2021-05-23:
I am testing an RT6 in my 307, it seems like it is working. Unfortunately I don't have a GPS antenna for it, so the navigation I can't try. But messages, trip computer, water and oil temperature works.

bronisMateusz, 2023-01-16:
@morcibacsi Did you test the RT6 with the large display or the Type C? I wonder if it would be possible to combine the RT6 and the Type C display and display the navigation directions as it was in the 206 with the RT1 or RT3.
[image]

morcibacsi, 2023-01-16:
@bronisMateusz The software doesn't support mixed setups as that would be an insane amount of work (if it was even possible). Every headunit works with its own display and that's it. So the RT6 works with its 7 inch display. The RT3 monochrome works with its type C display, but not with the newer type C which was meant to work with RD4/RD43/RD45. This is why the type C won't be supported with RD3. It's too much work, and doesn't worth it.

closed 2024-01-10 by morcibacsi