morcibacsi/PSAVanCanBridge #23: Compatibility question

Marc--, 2021-08-04:
Hello,

Thank you for your amazing work !! It'll help me a lot with my project, but a have some questions if you have time to answer it.

I own a 2004 VAN 307 (Navtech). My EMF Type CT was dead, so i remove it and the RT3. 
Now, i use a non PSA stereo, and can't read the car messages anymore (fuel autonomy, open doors, engine problems etc...).

My plan is to buy an EMF Type C screen, build your converter, and hook it up on the VAN network.

Did i miss something ? As i don't use a PSA stereo, i think i won't be able to configure the screen.

I'll now read again the hardware part and try to find the components !

morcibacsi, 2021-08-04:
Hi,

You are right, without a PSA headunit, you won't be able to setup the display. I have another project which hosts a webpage on the ESP32. You can connect to it with your smartphone and adjust the date and time, change between the display modes. You can find it here: [https://github.com/morcibacsi/PSAWifiDisplayControl](url) 
In theory it is possible to merge the two projects into one, but haven't had the time to work on it yet.

Marc--, 2021-08-09:
Thanks for your reply. If i'm correct, even without the PSA headunit, i'll be able to see the messages on the display, as they are send on the VAN network by the BSI (i don't know if it's the good name in english).

I've just received a type C screen, PCB's are on the way as the smd components. I'll also check your "configurator" for the screen.

Thanks for all your work.

morcibacsi, 2021-08-10:
Yes, there is no need for the original headunit or display. The messages are there, so having my board and the new display alone works.

Marc--, 2021-09-06:
Hi there, i finaly found some 10k smd in my garage ! Everything is wired and, great news, no smoke on first test !

But the screen stays black. I'll check wiring again, mine is a little different as my 307 is a "Navtech" wich was equiped with a RT3 (Fakra Quadlock connector on car side).

If i power the screen only with GND and +12v, is it supposed to turn on, or do i need to send something on the CAN pins, with your PSAWifiDisplayControl for exemple ? (I forgot to order a SN65HVD230...)

morcibacsi, 2021-09-11:
Hi,

Yes, it does need a wake up signal. For a quick test you can use this project:

[PSAWifiDisplayControl](https://github.com/morcibacsi/PSAWifiDisplayControl)

Make sure to uncomment this line, so the board sends the required messages.
https://github.com/morcibacsi/PSAWifiDisplayControl/blob/master/Config.h#L21

Marc--, 2021-09-17:
Hi, finished the soldering of the second board. This time only the minimal components ! And it works ! Thank you a lot ! 

Do you have a paypal adress ? I'll like to send you a little something for the time you spend and the sharing of this project !

Some minor changes, the step down converter died, so i replace it with a LM7805 that was laying around.

Now i will try to find a way to add the "Wifi controler" programto this board, as all the 4 SN65HVD230 i bought are defective :/

My goal is to activate wifi for the first 5 minutes, so i can configure the screen just after powering the board, and then cut it.
[image]

morcibacsi, 2021-09-17:
Thank you for the update, glad you suceeded :) It is always nice to see when people implement my project. My paypal address is the same as my email address (pinterpeti@gmail.com), however I didn't make the project for profit. But of course if you feel like donating, it is welcome :D It gives motivation to keep improving the project. 

Unfortunately those SN65HVD230 transceivers aren't that reliable. At least in my experience from 10 pieces there are more defectives than working ones. In later designs I replaced them with MCP2551 and TJA1040/TJA1050 and the like. 

Creating a wifi controller inside this project is a long-planned feature, I just haven't had the time to implement it :( So feel free to experiment with that, as currently I am quite busy with real life matters.

Chaozzmax, 2021-09-20:
> Hi, finished the soldering of the second board. This time only the minimal components ! And it works ! Thank you a lot !
>
> Do you have a paypal adress ? I'll like to send you a little something for [...]

i'm waiting for this additional program :-)

closed 2023-01-21 by morcibacsi