morcibacsi/PSAVanCanBridge #27: RMT RX buffer full

nickfarmer, 2022-11-30:
Hi Peter, This is a great project. Thanks for making it available to all. I have built a board (v1.4) and made up a wiring harness, but I am not quite able to get the VAN/CAN bridge working. Vehicle is a 2004 Citroen C3.

The ESP32 is stuck in a rebooting loop, and prints the message RMT RX BUFFER FULL (more complete output attached).

[ESP32OUTPUT.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/10122426/ESP32OUTPUT.txt)

Is this something you have seen before, or do you have an idea what my issue could be?

I have left Config.h as it is in the repo, and I have changed platformio.ini to move lib_extra_dirs into the env: section (otherwise the compiler returns an 'unknown configuration option' error) and to set the serial baud rate to 115200. I get a warning during compilation that driver/can.h is deprecated and that I should use driver/twai.h instead (attached below). I haven't tried to change anything to get this error to disappear. Could this be linked?

[compiler_output.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/10122619/compiler_output.txt)

Thanks,
Nick

morcibacsi, 2022-11-30:
Hi Nick,

Please try this sketch first for VAN monitoring, to make sure that there is no hardware issue: https://github.com/morcibacsi/esp32_rmt_van_rx/issues/9#issuecomment-1044846019

Also important to use the minimal version of the hardware, that works for sure (I use that version for years now).

I will look into the compiler warnings/errors soon.

nickfarmer, 2022-11-30:
Thanks for replying so promptly.

I tried sketch that you linked to: Output is 'CRC ERROR!' repeated. I think this points to a hardware problem.

I built the minimal v1.4 version using the absolute minimum component list. I've attached a photo to be sure.

I have tested the continuity between all of the pins that are supposed to be connected (e.g., U2 pin 1 to TSS463 pin 12). Do you think this might be a defective MCP2551?

[image]

morcibacsi, 2022-12-04:
Either a defective MCP2551 or some other hardware related error. What you can try is to wire the VAN DATAB pin directly to HV4 (bypassing the transceiver) to check if you receive anything. Hmm... if it doesn't work also try the other VAN pin I can't remember from my top of my head, but one of them should give you good results if everything is OK

[image]

closed 2023-01-21 by morcibacsi

BlayeeR, 2023-05-11:
Hey Peter
I built minimal version of the hardware and I got the same error message as in the title.
Checked example sketch and at the beginning Im getting 'CRC ERROR' but after a few of these I see data coming in without any errors.

Worth noting that the board does that only when its connected to the car and unit is in place. 
When I power it up without connecting, it does not restart.
I can see that its able to send CAN messages, because if I keep on clicking 'turn on' button on the unit, then sometimes screen gets lit for a flash of a second. It also means that it can read VAN, because it needs ignition frame for the screen to be able to turn on, right?
I managed to get it to run for a few second twice, but then it restarted again. 
Already swapped both of my MCP2551 but nothing changed, unfortunely I dont have replacement for TSS463 to test.

Do you have any idea how can I debug it or troubleshoot to find the issue? 
Already tried connecting HV pins directly to data pins but the behaviour didnt change.

Later when I will have more time I will try to do all above tips once again, also will try to solder everything onto second board as I got them in batch of five. But for now if you have any more tips, then it would be highly appreciated.

morcibacsi, 2023-05-15:
Some basic functionality should work without the TSS463. Maybe try replacing the voltage regulator, or try powering the board with an USB cable from the ESP32 (but make sure that in this case the +12V is not connected)

BlayeeR, 2023-05-17:
Yesterday I did some tests.

So I connected board to the car without 12v pin and via micro usb to my laptop, so that it uses power from the laptop.
When I started the unit from the button it worked, screen turned on. I could pair the vin, listen to music etc. but things like temperature were not working. When I turned the ignition on, then the board started restarting again.

At this point I wanted to give up, but went back to the car to check the logs. I had to turn off bt serial for that, because with board restarting bluetooth keeps disconnecting. 

With `//#define USE_BLUETOOTH_SERIAL` commented out board started to work, it reads temperature, trip data, works with ignition on.
Connected 12v pin back to power it up from the car and it kept working like that.

I played around with `Config.h` and using 
`#define SEND_AC_CHANGES_TO_DISPLAY`
with 
`//#define USE_NEW_AIRCON_DISPLAY_SENDER`
commented out also caused board to be stuck in infinite loop.

So both USE_BLUETOOTH_SERIAL and old aircon display sender cause board to fail. Ofc it might be not related closely with these settings- looking at how unstable it behaves, and anyway its 100% hardware issue and not the implementation.

It now works partially so I dont think voltage regulator is the problem.
There is one more thing that gets me to believe its TSS463 thats causing all the trouble. I disconnected old display and as expected AC and trip data stopped to work. So put both jumpers onto the connector pins. AC messages started to work(not sure if all correctly) but trip data is still not working. According to the readme it works only with TSS463 build, so I think I will buy another chip from diffrent seller and try replacing it
Maybe will try also with diffrent ESP32 board to make sure its fine too.

Overall Im happy, because I can already use it even if its not fully working. Will keep this thread updated, because there is not that much info about this issue

BlayeeR, 2023-05-17:
Whats the diffrence between TSS463 and TSS463C. Is it only diffrence in revision/mouting type/size etc. or it has some significant diffrences that could matter in that case?

morcibacsi, 2023-05-17:
It is nice to hear that you made it (partially) working. For the software it doesn't matter whether you install a TSS463 or a TSS463C, it is just a revision number, both should work fine.