morcibacsi/PSAVanCanBridge #28: compilation failed -- platform.IO  and Arduino IDE

JozimarHM, 2023-01-21:
Both in the compilation on the arduino ide platform and on the platform.IO there was an error in the compilation, on the platform .IO I solved it by commenting the line that generated the fault: PSAVanCanBridge/src/SerialPort/BluetoothSerialAbs.cpp: In member function 'virtual void BluetoothSerAbs:: begin(long unsigned int)'

It has something to do with the divergence of the VOID and INT types passed to the function.
On the Arduino it is:
error: no matching function for call to 'ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX()'
          _van_rx = new ESP32_RMT_VAN_RX();
 
But the function is there.

[image]
[image]
[image]

morcibacsi, 2023-01-21:
I fixed the bluetooth library compilation error. The esp32 team changed the function signature in one of their updates. 

You shouldn't happen to have the error related to ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX() as in platformio.ini it is defined that only allow the 1.x versions of the library. I rewrote that library and the software wasn't updated to use the latest version (2.x) of it, but is shouldn't really matter as platformio should use 1.x. So make sure you are using 1.0.0 or 1.0.1 and not 2.0.0

closed 2023-02-26 by morcibacsi

bronisMateusz, 2023-12-03:
@morcibacsi Hi, today I tried to compile the program at my place and had an identical error to the above on version `2.0.0` of the `ESP32 RMT Peripherial VAN bus reader` library. I changed it to `1.0.1`, as per your suggestion in the previous comment, but now I have the following error:

`In file included from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/PSAVanCanBridgeMain.cpp:17:
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h: In constructor 'VanMessageReaderEsp32Rmt::VanMessageReaderEsp32Rmt(uint8_t, uint8_t, IVAN_LINE_LEVEL, IVAN_NETWORK_TYPE)':
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:22:168: error: no matching function for call to 'ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX(const uint8_t&, uint8_t&, uint8_t&, VAN_LINE_LEVEL, VAN_NETWORK_TYPE)'
         _van_rx = new ESP32_RMT_VAN_RX(VAN_DATA_RX_RMT_CHANNEL, rxPin, ledPin, static_cast<VAN_LINE_LEVEL>(vanLineLevel), static_cast<VAN_NETWORK_TYPE>(vanNetworkType));
                                                                                                                                                                        ^
In file included from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:13,
                 from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/PSAVanCanBridgeMain.cpp:17:
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:37:5: note: candidate: 'ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX()'
     ESP32_RMT_VAN_RX();
     ^~~~~~~~~~~~~~~~
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:37:5: note:   candidate expects 0 arguments, 5 provided
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:29:7: note: candidate: 'constexpr ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX(const ESP32_RMT_VAN_RX&)'
 class ESP32_RMT_VAN_RX
       ^~~~~~~~~~~~~~~~
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:29:7: note:   candidate expects 1 argument, 5 provided
In file included from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/PSAVanCanBridgeMain.cpp:17:
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h: In member function 'virtual void VanMessageReaderEsp32Rmt::Receive(uint8_t*, uint8_t*)':
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:27:18: error: 'class ESP32_RMT_VAN_RX' has no member named 'ReceiveData'; did you mean 'Receive'?
         _van_rx->ReceiveData(messageLength, message);
                  ^~~~~~~~~~~
                  Receive
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h: In member function 'virtual void VanMessageReaderEsp32Rmt::Init()':
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:32:18: error: 'class ESP32_RMT_VAN_RX' has no member named 'Start'
         _van_rx->Start();
                  ^~~~~
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h: In member function 'virtual void VanMessageReaderEsp32Rmt::Stop()':
/Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:37:23: error: no matching function for call to 'ESP32_RMT_VAN_RX::Stop()'
         _van_rx->Stop();
                       ^
In file included from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/src/Van/VanMessageReaderEsp32Rmt.h:13,
                 from /Users/mateuszbronis/Desktop/PSAVanCanBridge/PSAVanCanBridge/PSAVanCanBridgeMain.cpp:17:
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:42:10: note: candidate: 'void ESP32_RMT_VAN_RX::Stop(uint8_t)'
     void Stop(uint8_t channel);
          ^~~~
/Users/mateuszbronis/Documents/Arduino/libraries/ESP32_RMT_Peripheral_VAN_bus_reader_library/src/esp32_arduino_rmt_van_rx.h:42:10: note:   candidate expects 1 argument, 0 provided

exit status 1

Compilation error: no matching function for call to 'ESP32_RMT_VAN_RX::ESP32_RMT_VAN_RX(const uint8_t&, uint8_t&, uint8_t&, VAN_LINE_LEVEL, VAN_NETWORK_TYPE)'`

morcibacsi, 2023-12-03:
> @morcibacsi Hi, today I tried to compile the program at my place and had an identical error to the above on version `2.0.0` of the `ESP32 RMT Peripherial VAN bus reader` library. I changed it to `1.0.1`, as per your suggestion in the previous comment, but now I have the following error:
> UPDATE: Through platform.io it went without a problem.

@bronisMateusz  Glad you made it work. In general I recommend using platform.io, With that these library version mismatches can be avoided totally. The same problem occurs also with the board definitions when esp32 releases a new esp32-arduino framework. From a developers perspective it is impossible with the Arduino IDE to make code which compiles for everyone, due to the million possible library and framework combination. In platform.io the board and library versions are locked in the platformio.ini so during compilation I can make it sure that the correct libraries will be downloaded.