morcibacsi/PSAVanCanBridge #32: some issues...

MRWZJP, 2023-11-20:
hi! thank you for the project! i made this V2C hardware for my peugeot 206 and connect the wire correct. its not working fine,it can be turn on to use audio and bluetooth,but trip computer data all display 0,evev the tempture number also zero and its flashing.i think maybe wire is wrong,is the DATA/DATAB =VAN+/VAN- ? i find some V2C installation work, no one connect the VANbus on the ISO-A connector. peoples make the VANbus connect to dashboard's 5 and 10 pins and its working.

morcibacsi, 2023-11-21:
It doesn't matter where you are connecting to the VAN bus. The 5/10 pins on the odometer connects to the same network as the 2/3 pins on the radio (both are labeled as 9004/9005) on the schematics:
[image]

I don't know what are you referring to as VAN+ and VAN- I never used that name for the VAN data lines. But if you check the schematics here, https://github.com/morcibacsi/esp32_rmt_van_rx you can see that DATA should be connected to the CAN LOW pin on the transceiver, and /DATA (or DATAB) should be connected to the CAN HIGH pin on the transceiver.

DATA = 9004 = Pin 5 on odometer = Pin 2 on the radio
DATAB = /DATA = 9005 = Pin 10 on odometer = Pin 3 on the radio

I suggest to double check your wirings and maybe run the basic VAN monitor sketch to see if that works:
https://github.com/morcibacsi/esp32_rmt_van_rx/blob/master/examples/esp32_arduino_van_monitor/esp32_arduino_van_monitor.ino

MRWZJP, 2023-11-25:
Maybe it's a problem with the sales market, and now I fix this problem，THX
[image]

MRWZJP, 2023-12-11:
Hello, I have been using it for some time now and have found some issues. When you are driving normally, 1:some times the USB song will automatically switch, 2: Bluetooth will disconnect, and reconnect after few secs, 3:the mileage counter on the dashboard will be cleared. I think it may be caused by a bad connection in the circuit, but the above problem still exists after I have soldered all the circuits.and i already make two of this hard ware,also have issuses.one of LM7805,one of DC-DC moudle.

morcibacsi, 2023-12-12:
I never faced such issues, so I don't have any useful ideas.

bronisMateusz, 2023-12-14:
I, too, am having strange mounting problems with a 2004 206 RC. After plugging the kit in, everything works for about a few seconds, then you hear some sounds (like relays) then the cluster in front of the driver stop working (they don't indicate anything, the lights flash). @MRWZJP did you maybe change something in the way of connection relative to the original guide?
[image]

morcibacsi, 2023-12-15:
@bronisMateusz Can you show me a picture of your board? Also if you try the receiver sketch from here https://github.com/morcibacsi/esp32_rmt_van_rx do you face the same issues?

bronisMateusz, 2023-12-18:
@morcibacsi Sorry it's only now, but I've sent my kit for diagnostics to @karolp1993, because he has a kit with BSI from the C4 on his desk and interestingly enough, everything worked fine at his place.

Here pictures of the board itself:
[image]
[image]

I have the wiring harness all done exactly as it was on yours. I didn't connect the pins together in the connector from the old display, because in the 206 the air conditioning works fine without that.

karolp1993, 2023-12-18:
I have C4 BSI for coding display/radio and check it natively. I checked Mateusz's adapter on my bench with BSI from 307 and everything works fine for few days. In 206 display and aircon panel are separetly connected to instrument cluster. But like Peter mentioned, it should not be te case due this connections just connect to the same bus. 
[image]

bronisMateusz, 2023-12-21:
@morcibacsi I did a quick test of the latest version from `main` branch. **A piece of amazing work!** The board behaves stably, the status messages display correctly, the only thing that doesn't work is the radio control stalk from steering wheel.

morcibacsi, 2023-12-21:
@bronisMateusz 

The new version has a wifi access point where you can set up a few things. Make sure you enable C3. A more in-depth documentation for this is in the works.

[image]