morcibacsi/PSAVanCanBridge #34: BSI crash after few minutes

bronisMateusz, 2024-01-06:
Hello,

I am still struggling with the configuration of the whole project in my Peugeot 206 RC and code from `main` branch.

I replaced the display from color to monochrome (I wanted BT support, and the color one didn't offer it without an additional module). Before that, I didn't have a chance to test the project longer, because I was waiting for this display.

Today I managed to do it and unfortunately I still have problems with the BSI. After a few minutes of running the kit, the BSI stops working, the main gauges  start flashing.

Display number: `9803399680`
[image]

When everything is connected, some of the options from the display are unavailable:
[image]
(I don't know if it should be possible to do this, but you can't highlight that first icon.)
[image]
[image]

In the settings of the board I have: **C3 - activated, C11 - activated, S1 - RD45, S2 - CAN2004, E2 - 60**

morcibacsi, 2024-01-06:
Finding a color display with Bluetooth and USB support is almost impossible. It requires a very high version number. Sw revision above 60 (6x.xxx or something like that). Maybe the ref number 9667027580 is good, but this isn't 100% accurate info.
 
The first icon is only available when the radio is turned on as I remember. 

The "Define the vehicle parameters" is always disabled as there is nothing you could set up on a car with a VAN bus (there would be settings to enable DRL, parking assistant, follow me home light and stuff like that, but those aren't available on older cars). So even if I'd enable the menu, it wouldn't do much. 

The "Manage the telephone call" is only active during a phone call as I remember.

Regarding the BSI crash you can try enabling C1 so my board won't write anything to the VAN bus. Did you remove the original display? If you enable C1 and you don't have the original display the consumption info and the door status won't work, that is perfectly normal. So try enabling C1 and reconnect the original display, and check if the BSI still crashes.

bronisMateusz, 2024-01-07:
Thank you so much for explaining everything. Actually leaving the old display in the 206 and enabling the C1 option solves all the problems. I did some miles today and there were no problems.

morcibacsi, 2024-01-07:
Thanks for the feedback. I would like to fix this issue, and I would be glad if you could help me as I can't reproduce this issue on my car. Please capture some VAN bus logs for me if possible with the original display attached, and driving around for a few minutes. If you need further instructions I am happy to provide.

bronisMateusz, 2024-01-07:
Sure, I will try to register something this week.
I understand that I need to connect via bluetooth the computer and ESP32, and read the logs in PlatformIO?

morcibacsi, 2024-01-07:
**VAN log via bluetooth**

If you comment out this line: https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/Config.h#L9
and remove the comment from the start of the next line: https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/Config.h#L10
then bluetooth will be started instead of wifi. You can follow this guide on capturing with a phone: https://github.com/morcibacsi/PSAVanCanBridge/blob/master/wiki/debugging.md#3-van-bus-capture--software-replay
The log can be saved to a file in the ... menu on the right by clicking the Data/save item.

**VAN log via USB**

If you don't have an Android phone but you have a laptop, then you can connect with an USB (make sure you don't connect the 12V from  the car in this case). Also in this case you don't need to modify the config,h file, you need to use the wifi version.
See on how to open the serial monitor in Arduino here:
https://forum.arduino.cc/t/using-the-serial-monitor-for-debugging-analysing-what-is-going-on-inside-your-code/978443
(make sure you select the correct COM port, and enable the "Show timestamps" checkbox)
The serial monitor output can be copied to a text file.

The best would be to have two logs, one with the original display connected and C1 enabled (1-2 minutes is enough, you don't need to drive, just make sure you open and close the door a few times). And another log with C1 disabled, and the original display removed (same method as the previous, maybe until the BSI crashes). This way I can try comparing the original behavior and my version, and try figuring out what is the difference which causes the BSI to crash.

Also I had an idea, regarding the crash, it is a shot in the dark, but try increasing this value to 500 and disable C1 (and remove the original display to see if that fixes the issue)
https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/src/Van/Writers/VanDisplayStatus.h#L14

Thank you very much for your help and patience.

bronisMateusz, 2024-01-08:
Ok, thanks for all the information. I have the logs. The one withoutC1 is really extensive, because I waited until the BSI stopped working.

[logC1.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/13859341/logC1.txt)
[logWithoutC1.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/13859343/logWithoutC1.txt)

Yesterday I also did another longer route (1.5 hours) and at the end I experienced strange behavior of the display and radio. They stopped working for a while. They would then turn on for a bit and turn off again (for longer and longer periods). This was a configuration with C1 turned on and the original display connected.

morcibacsi, 2024-01-08:
Thanks for the log files. I analyzed them, and maybe I found something. To display the door status and consumption data the following happens: the BSI sends a message (lines starting with 0E 8C 4C) to the display that there is new data to show. When the display receives this message it sends a request (lines starting with 0E 56 4E) to the BSI for the new data. 56 4E is an awkward message as half of it is from the display, the other half is from the BSI.
With the original display it seems these messages are go side-by-side. So after a 8C 4C is received, a 56 4E follows immediately, and after that 8C 4C is received a bit later.
But with my code after I receive a 8C 4C even though I send the 56 4E almost immediately, the BSI keeps sending the 8C 4C, like if it wouldn't register my query with 56 4E.

I suspect I don't send the 56 4E "fast enough" after  8C 4C and the BSI keeps sending the 8C 4C again, and again. And maybe that causes the crash.

So... I would ask another pair of logs if you don't mind, where I can see the timestamps of the messages to verify this. As I saw you are using platformio on linux (?) I think you need to add monitor_filters = time to platformio.ini under monitor_speed to enable timestamps.

> Yesterday I also did another longer route (1.5 hours) and at the end I experienced strange behavior of the display and radio. They stopped working for a while. They would then turn on for a bit and [...]

I think this is caused by the overheating of the LM7805. I am going to modify the software so the Wifi hotspot won't be enabled all the time. That way the board won't draw that much current, so the LM7805 won't get overheated.

Drsimson, 2024-01-10:
Hello sir, it is good that this project is alive after a long time 👌.  The structures are now more complete, but there are some problems.  Mode 2 does not display speed and RPM.  Wi-Fi configuration works on old devices and new devices cannot load the page.

morcibacsi, 2024-01-10:
> Hello sir, it is good that this project is alive after a long time 👌. The structures are now more complete, but there are some problems. Mode 2 does not display speed and RPM. Wi-Fi configuration works on old devices and new devices cannot load the page.

Please open a separate issue.

Drsimson, 2024-01-10:
> > Hello sir, it is good that this project is alive after a long time 👌. The structures are now more complete, but there are some problems. Mode 2 does not display speed and RPM. Wi-Fi configuration works on old devices and new devices cannot load the page.
> 
> Please open a separate issue.

Ok

bronisMateusz, 2024-01-14:
Sorry it took so long, but I was busy during the week. I'm uploading more logs from today.

I'm using PlatformIO on Mac OS and had to add in `platformio.ini` exactly this line: `monitor_filters = time`.
(this is for people who would like to do something like this someday too)

As for the logs themselves, this `logWithoutC1` is very extensive, as BSI work surprisingly well for a long time. It was only when I tried to reset the mileage by holding down the button on the right stalk (the one from the wipers) that the BSI stopped working.
[logC1.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/13931527/logC1.txt)
[logWithoutC1.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/13931528/logWithoutC1.txt)

> I think this is caused by the overheating of the LM7805. I am going to modify the software so the Wifi hotspot won't be enabled all the time. That way the board won't draw that much current, so the LM7805 won't get overheated.

As for overheating, my board looks like this:
[image]
Karol mentioned to me that for lowering the temperature I should use a voltage converter. Maybe this also affects this problem.

morcibacsi, 2024-01-29:
Sorry for the late reply, I was busy with other stuff. Thanks for the additional logs, I saw the same as before 8C 4C is sometimes keeps getting sent, but I am not really sure if that is a problem.
However indeed it is strange that this time it took longer before it crashed. I am working on a newer hw revision where the overheating issue will be resolved, and maybe thee TSS463 also won't be needed anymore. It will definitely take a few weeks to finish it, so please stay tuned.

bronisMateusz, 2024-01-29:
No problem, I know that such things can take some time. Can I still provide you with some additional data? A few comments earlier you mentioned:

> Also I had an idea, regarding the crash, it is a shot in the dark, but try increasing this value to 500 and disable C1 (and remove the original display to see if that fixes the issue)
https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/src/Van/Writers/VanDisplayStatus.h#L14

Would you still like to check it out?

morcibacsi, 2024-01-29:
Well, you can give it a try, just to check all possibilities.

morcibacsi, 2024-02-28:
> I think I’ve got the same issue. What is the year of your car @bronisMateusz ? Mine is a 2002 model. Maybe it’s related ?
> 
> I’ve just installed the bridge in my car, with a RD45 and a new screen. I had to compile the main branch as the master didn’t work and made my bsi crash instantly. I had issues with the compilation of the main branch, as some of the includes are wrong (some includes of Config.h). After that, it worked, but after 10 minutes of tests, the bsi crashed again. I had to unplug the battery of the car.
> 
> Can I help with something ?

@Toniob Thanks for reaching out. The main branch is much more stable, and behaves more like the original display, so you should definitely use that.
I spent a few hours trying to make some sense of various VAN log files, but I couldn't see any obvious clues (apart from what I mentioned earlier). And (of course) I can't reproduce it in my car (but it is a 307 from 2004). This issue is so annoying. Unfortunately I am currently out of ideas on how to solve this. Except the workaround to hide the original display somewhere and enable C1 in the settings. If you could somehow solve it / debug it further that would be nice. And of course if you have any questions regarding the inner workings of the software, feel free to ask.

Toniob, 2024-02-28:
Would it help if I send you some logs ?
And I don’t know if it’s related, but I didn’t manage to stop the thief protection. I’ve put my VIN number in the web interface and did the konami like code. Is it how it’s supposed to be done ?

bronisMateusz, 2024-02-28:
@Toniob 

> What is the year of your car @bronisMateusz ?
Mine is from 2004.

If you are able to send logs, then you can check the solution that @morcibacsi mentioned. I, unfortunately, have not been able to log them recently. I will have the car again in about 3 weeks:

>  Also I had an idea, regarding the crash, it is a shot in the dark, but try increasing this value to 500 and disable C1 (and remove the original display to see if that fixes the issue)
> >https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/src/Van/Writers/VanDisplayStatus.h#L14

morcibacsi, 2024-02-28:
> Would it help if I send you some logs ? And I don’t know if it’s related, but I didn’t manage to stop the thief protection. I’ve put my VIN number in the web interface and did the konami like code. Is it how it’s supposed to be done ?

Yes, if possible send some logs when the BSI crashes. Maybe I can figure out something.

Toniob, 2024-02-28:
Thank you for your answers. I’ll try the auto-pair button tomorrow. How can I know if I need to use CAN2004 or CAN2010 ? Same question for the radio type. Should I use RD45 or RD45 2010 ?

And by the way, thank you for your awesome work.

morcibacsi, 2024-02-28:
> Thank you for your answers. I’ll try the auto-pair button tomorrow. How can I know if I need to use CAN2004 or CAN2010 ? Same question for the radio type. Should I use RD45 or RD45 2010 ?
> 
> And by the way, thank you for your awesome work.

CAN2004 (or AEE2004 to be precise) was used in older cars made until ~2012. After that PSA modified their CAN protocol (CAN2010/AEE2010) where some messages have been removed some changed and got new meaning. 

If you have one of the 12 pin type displays, then you need to use CAN2004:
[image]
[image]

If you have the 6 pin display, you need to use CAN2010:
[image]

Take care that you can't mix AEE2010 and AEE2004 peripherals. So if you choose the 12pin display then it will only work with the "old version" of the RD4/RD43/RD45 which worked with the AEE2004 protocol.

morcibacsi, 2024-02-29:
@bronisMateusz  @Toniob Please send me a picture of the sticker on your BSI. Maybe I can buy the same type and try reproducing the issue on my bench.

Toniob, 2024-02-29:
Here are 2 dumps. I’ve tested 2 different screens. Type C both of them, but not the same year, and the color are different. The BSI crashed the 2 times, quite fast (2 minutes tops). C1 was not activated and I didn’t have the old screen plugged in.

[serial_20240229_103935.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/14446132/serial_20240229_103935.txt)
[serial_20240229_102926.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/14446133/serial_20240229_102926.txt)

And here is a picture of the sticker
[image]

Toniob, 2024-02-29:
I’ve tested with C1 activated and it didn’t crash (I didn’t have to old screen attached). So it really seems it related to the VAN packets which are sent.

morcibacsi, 2024-02-29:
Yes, definitely the sending is the problem, unfortunately it is required to have the trip computer working. So if C1 is activated and there is no original display installed, then the door statuses and the trip computer won't be available.

Toniob, 2024-02-29:
Right now, I can live without the trip computer, that’s not a real issue for me. I can wait.

Do you think you can find the same BSI as I have ? Does @bronisMateusz have the same one ?

I’ve realized that my dumps are small. It didn’t put the buffer to unlimited. Do you want me to do them again, or you’ve got what you needed in them ?

morcibacsi, 2024-02-29:
The dumps are fine. I managed to find a BSI, it will arrive next week. In the meantime I have an idea, to reduce the number of VAN sending and see if that helps. Please comment out this line to see if that changes anything.
https://github.com/morcibacsi/PSAVanCanBridge/blob/main/PSAVanCanBridge/src/Van/Writers/VanDisplayStatus.cpp#L7

I bought this BSI: https://allegro.pl/oferta/peugeot-206-modul-skrzynka-bsi-9646777380-europa-14608567084

morcibacsi, 2024-02-29:
Ehh.... on second though, please repeat the logging again.

Toniob, 2024-02-29:
> I bought this BSI: https://allegro.pl/oferta/peugeot-206-modul-skrzynka-bsi-9646777380-europa-14608567084

The number starting with S is a little different with mine. It ends with an E and not a G. But I don’t know if that’s a big difference.

I’ll try commenting the line you asked and test again soon. And I’ll make better logs this time ;)

morcibacsi, 2024-02-29:
> The number starting with S is a little different with mine. It ends with an E and not a G. But I don’t know if that’s a big difference.
> 
Yeah, I saw that, couldn't find an exact match. Let's hope for the best.

morcibacsi, 2024-02-29:
Oh and btw we have a nice community around PSA related development here, if you like joining. Everyone is welcome: https://discord.gg/DPthrN2cbS

morcibacsi, 2024-02-29:
Sounds promising 🙂 that message is required for the trip computer reset, but I can rewrite the code to send it a few times when the actual reset happens. Please test it for a few days, and we'll see what to do next.

Toniob, 2024-03-06:
> PS: For the time being, I removed the superfluous components from the original B-type display I had and hid it behind the C-type display to keep it out of my way. This solution seems to be working at the moment. I also once spoke with a person who uses this design in Iran, and they also all do it this way for the 206.

Hello @bronisMateusz . Do you have a description of the superfluous components you removed and how you did it ? Thank you

bronisMateusz, 2024-03-16:
@morcibacsi I did a test today with increasing the value for the `SEND_STATUS_INTERVAL` variable with the original display disconnected and C1 set to false. Everything seems to work correctly, nothing bad happened. Here are the logs:
[log500WithoutC1.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/14624257/log500WithoutC1.txt)

> Hello @bronisMateusz . Do you have a description of the superfluous components you removed and how you did it ? Thank you

I was a bit imprecise in my words. I simply removed the plastic casing from the display and placed it under the visor so that it could be put on easily:
[image]
[image]

Toniob, 2024-03-16:
Thanks for you answer. I bought a used emf module, and I already did you’ve shown. I even removed the screen, to gain more space. And it’s working fine (and the car goes to sleep).

morcibacsi, 2024-03-16:
> @morcibacsi I did a test today with increasing the value for the `SEND_STATUS_INTERVAL` variable with the original display disconnected and C1 set to false. 

What value did you use?

bronisMateusz, 2024-03-16:
500 as you suggested. I am driving the car right now and unfortunately the BSI has stopped working again

morcibacsi, 2024-03-16:
@Toniob @bronisMateusz 
Alright, I pushed some changes to the [bsi_crash_fix](https://github.com/morcibacsi/PSAVanCanBridge/tree/bsi_crash_fix) branch please check if that helps resolving the BSI crash and the failure to send the BSI to sleep

morcibacsi, 2024-03-16:
> @Toniob @bronisMateusz Alright, I pushed some changes to the [bsi_crash_fix](https://github.com/morcibacsi/PSAVanCanBridge/tree/bsi_crash_fix) branch please check if that helps resolving the BSI crash and the failure to send the BSI to sleep

Oh, and please also check if trip reset is still working. 

And of course without the original display present.

bronisMateusz, 2024-03-17:
Ok, I ran tests today and strange things are happening. I tested for less than 15 minutes, including while driving and the BSI still worked properly. The button on top of the windshield wiper handle does not work. You can't switch between screens with it. Also, the statuses from the open doors don't work, nor the fuel consumption display.

I'll wait an hour or so, let the BSI sleep, and try to take another logs.

[image]

morcibacsi, 2024-03-17:
> Ok, I ran tests today and strange things are happening. I tested for less than 15 minutes, including while driving and the BSI still worked properly. The button on top of the windshield wiper handle does not work. You can't switch between screens with it. Also, the statuses from the open doors don't work, nor the fuel consumption display.
> 
> I'll wait an hour or so, let the BSI sleep, and try to take another logs.
> 
> ![IMG_0386](https://private-user-images.githubusercontent.com/45051768/313457220-fd1005da-697f-4d51-8088-0ca4304cd9d4.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2NjcxMTYsIm5iZiI6MTcxMDY2NjgxNiwicGF0aCI6Ii80NTA1MTc2OC8zMTM0NTcyMjAtZmQxMDA1ZGEtNjk3Zi00ZDUxLTgwODgtMGNhNDMwNGNkOWQ0LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxN1QwOTEzMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ODZmMzhjNWRjOWEyMmQ4NTc2YWVlNzU5MDM2MTZkOTZjMGY1OTljYWFiOTVlYzcwMWYxYWMxZjRkN2FkMWY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RVcoBJwoW_-Xc0aKI6ohatpLUu0VBl3Po5oXcWPZ6JM)

Unfortunately I am not near my car, so I can't test the changes, but looking at the code the trip computer should work. Is there a chance that C1 was left enabled?

bronisMateusz, 2024-03-17:
Interesting case. I went to check the situation again. In the `config.h` file I had these changes:
```
bool BRIDGE_REMOTE_STALK_DATA = true;
bool USE_IGNITION_SIGNAL_FROM_VAN_BUS = true;
uint8_t FUEL_TANK_CAPACITY_IN_LITERS = 50;
bool ENABLE_MANUAL_GEARBOX_DISPLAY = false;
uint8_t TIRE_SIZE = 17;
uint8_t TIRE_WIDTH = 205;
uint8_t TIRE_RATIO = 40;
uint8_t RADIO_TYPE = 2;
```

I went via bluetooth into the adapter settings, and there C1 was still selected, the tank capacity was also different from what I had set before uploading the code:
[image]
[image]

In the attached C1 change file there are about 16,000 lines of code. After that the door statuses and combustion indications actually returned. Trip reset suddenly not working.

morcibacsi, 2024-03-17:
The settings from config.h are are just default values. As soon as you save the settings from the configuration page they are written to a file in the flash. On startup they are read from the file. So when the file exists changing the config.h has no effect (unless you delete it with the Delete config button). I try to check why the trip reset isn't working.

bronisMateusz, 2024-03-17:
Thanks for the clarification. I also forgot to include the second log file I mentioned in my previous message: 
[logBSICrashFix2.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/14627162/logBSICrashFix2.txt)

morcibacsi, 2024-03-17:
I fixed the trip reset.

bronisMateusz, 2024-03-17:
I did more than 100 km with the car today. Everything worked perfectly 😁

bronisMateusz, 2024-03-19:
Ok, I uploaded the latest version of the code, resetting the odometer actually works, but you have to hold the button for a long time, about 10 seconds, although looking at the code, it should be about 3 seconds.

This is not some big problem, I am writing this more for information.

https://github.com/morcibacsi/PSAVanCanBridge/assets/45051768/1be8ff0d-b497-47e3-9f2a-3cc3ccbb1f58

[log.txt](https://github.com/morcibacsi/PSAVanCanBridge/files/14653457/log.txt)

morcibacsi, 2024-03-21:
Damn, I was really hopeful that finally we made some progress. Please test it if you can, and in the meanwhile I will try to come up with something.

bronisMateusz, 2024-03-25:
I know, the whole thing is frustrating, so thank you all the more for having the will to keep looking for a solution to these problems.

I've had a couple of rides in the car recently and nothing bad has happened. I wondered what might have caused the BSI crash then and thought maybe it was that I had reset the trip data. I tried resetting the trip data today, waiting a similar period of time as I did then, but nothing bad happened. I have the logs from this trip, but I'm not attaching them this time because I don't think it's necessary.

Have you perhaps managed to guess why the trip reset button is not working as quickly as before?

@Toniob Are you able to check whether recent changes have also helped to solve problems in your car?

Toniob, 2024-03-25:
I didn’t have the time to test or to take my car. My main issue is that I live in a city, without a garage. It’s not easy to test or to make modifications on the car. And if something goes wrong, like the last time, it’s a pain to charge the battery. But maybe I’ll do it at my parents next week.

Toniob, 2024-04-02:
Good news on my side. I’ve tested the bsi_crash_fix branch yesterday. I made 2 trips of 20 minutes each, and the BSI didn’t crash. I’ve also tested the sleep mode and everything went well. Maybe I’ll try with a longer trip in the next few days, but for the moment, it seems I won’t need the original screen anymore.

bronisMateusz, 2024-04-02:
Unfortunately, there are again problems on my part. I have been using the car extensively in the last few days and two days in a row I had a BSI failure. One occurred after only 1.5h of driving and the other after just a few minutes. Yesterday I did another 1,5h tour and everything was ok.I will try to register something in the nearest days.

@Toniob In your car also reset trip performs after such a long time button press, as in my video?

Toniob, 2024-04-07:
On friday, I made 2 trips of 2 hours each, and everything went ok with the BSI. But after the first, after a few hours, when I came back to the car, the display was not powering on. I had to unplug and plug again the bridge to make it work again. I think that's the esp32 which crashed. Too much heat maybe ?

Is there a way to have a watchdog on the bridge ? And if it doesn't reply anymore, it is rebooted automatically ? Something implemented in the ESP32 maybe ?

I also tried the trip reset and strange thing, the button on the car did nothing. No trip reset, but the button had no function anymore, not even the statistics display, as it should be. I'll probably try again another day. It remember it was working beofre so I really don't understand what happened.

bronisMateusz, 2024-07-08:
Hi, I am back with the logs after this long break. To be honest, I had lost a bit of hope and opted for stability, i.e. driving with 2 displays plugged in, but nevertheless this topic kept me going. I managed to catch the moment again today that crash the BSI on the most recent version of the code from the `bsi_crash_fix` branch. I hope this will give new information for this problem.

[log08.07.24.txt](https://github.com/user-attachments/files/16132576/log08.07.24.txt)

morcibacsi, 2024-07-10:
Thanks for the log, I'll try to investigate. I didn't have much time lately to work on the project (any project actually), the BSI I bough to investigate the issue is just resting on my desk, waiting to wire things up :(

bronisMateusz, 2024-07-10:
I completely understand this. Also, I haven't uploaded my BSI version here before as I haven't had the car for a long time, but this is the `9650584680` version.

morcibacsi, 2026-09-10:
Hey guys!

I attached a preview version where I attempt to fix the BSI crash issue (again). This is for boards with the ESP32-C6 controller. So basically all after board version v1.6. I couldn't test it in my car yet, only with a BSI on my desk. 

For updating you can either follow this:

[https://github.com/morcibacsi/PSAVanCanBridge#uploading-the-firmware](https://github.com/morcibacsi/PSAVanCanBridge#uploading-the-firmware)

This: [https://github.com/morcibacsi/PSAVanCanBridge/blob/v3/wiki/update-firmware-from-a-release.md](https://github.com/morcibacsi/PSAVanCanBridge/blob/v3/wiki/update-firmware-from-a-release.md)

Or also can upload from here: [https://thelastoutpostworkshop.github.io/ESPConnect/](https://thelastoutpostworkshop.github.io/ESPConnect/)

[image]

Or via OTA if you are on a at least version v3.5.0

[bsi_crash_fix_esp32c6_v16.zip](https://github.com/user-attachments/files/32076884/bsi_crash_fix_esp32c6_v16.zip)

Toniob, 2026-09-11:
Can you tell us what you changed in it ? Did you manage to reproduce the crash on your bench ?

morcibacsi, 2026-09-11:
I have created logic analyzer dumps with the BSI and the original EMF, and compared their behaviour with the bridge. There were some minor differences, but with this fix, my bridge produces the exact same waveform as the EMF. This test was with a BSI which never crashed, but I assume that if the waveform is a 100% match, then the crash shouldn't happen. 

On the following picture I marked the differences with the red rectangle. The timings of that part are on the right side in microseconds. My original implementation completely missed that 1 -> 0 part, instead it was 1-1. Now I added it, but the timings might need minor adjustments. It has some tolerance so I don't need to match the exact timing, currently I am trying to narrow down the boundaries. From my experience anything between 7.5 - 8.x is fine, now sometimes I have 9.313 microseconds and sometimes it is 8.3. But 9.3 might be fine as well.

[image]

The workflow of the messages is the following:

BSI sends 0x8C4 EMF sets that 1->0 transition, then EMF sends 0x564.
My bridge didn't set the 1->0 transition, but sent the 0x564
When the 1-0 transition wasn't set, the BSI repeated 0x8C4 a few more times, before it gave up.

With the fix in place the BSI doesn't repeat 0x8C4. So this is why I'm hopeful that this should work. As I remember in previous crash losg I saw that repeating 0x8C4 - 0x564 chit-chat between the BSI and my bridge before the crash.

morcibacsi, 2026-09-15:
I just pushed a commit to the v2 branch for those who are still using the v1.4 revision of the board with the TSS463 IC. I’m really hopeful that the crash should be fixed with this version. The commit introduces the same bus behavior as the original display, which is why I expect it to work.

Toniob, 2026-09-16:
> I attached a preview version where I attempt to fix the BSI crash issue (again). This is for boards with the ESP32-C6 controller. So basically all after board version v1.6. I couldn't test it in my car yet, only with a BSI on my desk.

I’ve installed it on my bridge. I’ll try it as soon as I can. I’ll keep you posted.

[9 more replies]

closed 2026-10-01 by morcibacsi