morcibacsi/PSAVanCanBridge #40: Can Do sniffing by this bridge??

hossammorayan, 2025-02-02:
Hello Guys I have Question I want to do sniffing by obd port to know frame of thing like when press light button know frame to re send again things like in book (The Car Hacker's Handbook - Craig Smith) and in this video(https://youtube.com/playlist?list=PLNiFaO8hU1z0o_6DSxk-jcVAM3UCUR-pY&si=sWTDRaY_bmBRk-9t) can tell me how to do this and am surprised of my car is use van protocol my car is citreoen xsara 2002

morcibacsi, 2025-02-03:
VAN bus data is not exposed on the OBD connector, so unfortunately you can't do that. Sniffing is only possible by attaching wires to the data cables directly. Relatively easily available connection points are behind the radio for the CONF bus or at the "Driving school connector" on the BSI for one of the CAR bus. Although re-sending isn't that easy because VAN bus is fundamentally different from the CAN bus with its in-frame reply messages. Man-in-the-middle message modification might be an option but it is much more difficult to do than on CAN bus.

closed 2025-02-04 by hossammorayan

hossammorayan, 2026-09-02:
hello @morcibacsi because am still excited to write van frames  in below photo thats mean can send van frame ??
Scenario : am sniff van frame of hazard light and need send to hazard light on can this happen in newest version or still not avaible ??

[image]

morcibacsi, 2026-09-04:
Unfortunately that is not how it works. You can't send any data on the wires, unless you remove the original sender from the network, otherwise you are ending up with contradicting commands (you would have 2 peripherals, sending different commands at the same time)

hossammorayan, 2026-09-05:
1-you mean by ( You can't send any data on the wires, unless you remove the original sender from the network)
thats mean any periperal cant accept  frame from 2 sender and it will accept frame from licensed one as from bsi and it ignore outside one  in this situation is me as used outside hardware like bridge ?? 
2-you mean by (you would have 2 peripherals, sending different commands at the same time)
thats mean thing in car  for example hazard light each time send different frame mean is fixed frame or random frame ??

thanks for your help

morcibacsi, 2026-09-05:
Messages are usually sent repeatedly, for example every 100 ms. They are not sent only once when something changes.

Normally, one ECU is responsible for sending a particular message. You cannot simply add another device that sends the same message with different data, because then both the original ECU and your device would continuously send conflicting values.

The receiving ECU does not know that one sender is the "original" and the other is an external device. It just receives the messages.

So if you want to replace a message, you need to prevent the original message from reaching the receiver and send your modified message instead. This is one of the purposes of using a bridge between two buses.